duckietown/gym-duckietown #2: standalone.py

paulaurel, 2018-03-15:
When running the standalone.py script I get an error message, related to env = gym.make(args.env_name).

maximecb, 2018-03-15:
Please paste the complete error message you got.

paulaurel, 2018-03-15:
This is the error I received: 
Traceback (most recent call last):
  File "./standalone.py", line 18, in <module>
    env = gym.make(args.env_name)
  File "/home/daurel/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/daurel/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/daurel/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/daurel/Documents/Bachelorarbeit/DuckieSim/gym-duckietown/gym_duckietown/envs/simplesim_env.py", line 281, in __init__
    self.multiFBO, self.finalFBO = createFrameBuffers()
  File "/home/daurel/Documents/Bachelorarbeit/DuckieSim/gym-duckietown/gym_duckietown/envs/simplesim_env.py", line 97, in createFrameBuffers
    True
  File "/home/daurel/.local/lib/python3.5/site-packages/pyglet/gl/lib.py", line 105, in errcheck
    raise GLException(msg)
pyglet.gl.lib.GLException: b'invalid operation'

maximecb, 2018-03-15:
Are you on a mac?

paulaurel, 2018-03-15:
using a virtual machine on a mac

maximecb, 2018-03-16:
So the issue is because the OpenGL drivers on the mac don't support some of the features that I'm using (supported by nvidia drivers). Specifically the use of anti-aliasing in frame buffer objects. I'll try to make it work. I have a macbook at home that I can use to test it.

paulaurel, 2018-03-16:
That would be perfect thank you.

paulaurel, 2018-03-16:
I just tried running the standalone.py on a pc and I got the following error 

./standalone.py 
Traceback (most recent call last):
  File "./standalone.py", line 18, in <module>
    env = gym.make(args.env_name)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 14, in load
    result = entry_point.load(False)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2431, in load
    return self.resolve()
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2437, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/__init__.py", line 3, in <module>
    from gym_duckietown.envs.duckiebot_env import DuckiebotEnv
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/duckiebot_env.py", line 10, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type

paulaurel, 2018-03-16:
I just ran the standalone.py code on a pc running linux and encountered the following problem

./standalone.py 
Traceback (most recent call last):
  File "./standalone.py", line 18, in <module>
    env = gym.make(args.env_name)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 85, in make
    cls = load(self._entry_point)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 14, in load
    result = entry_point.load(False)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2431, in load
    return self.resolve()
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/pkg_resources/__init__.py", line 2437, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/__init__.py", line 3, in <module>
    from gym_duckietown.envs.duckiebot_env import DuckiebotEnv
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/duckiebot_env.py", line 10, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type

fgolemo, 2018-03-16:
@paulaurel: could you try this please:
https://stackoverflow.com/questions/43019951/after-install-ros-kinetic-cannot-import-opencv

fgolemo, 2018-03-16:
... because from what I see in the traceback you are starting your script in Python 3 but then it tries to import OpenCV for PY 2.7 and that obviously won't work.

paulaurel, 2018-03-16:
Thank you. I am now able to import OpenCV in python 3. Yet I still cannot run the standalone.py code. I now encounter the following error 

Traceback (most recent call last):
  File "./standalone.py", line 18, in <module>
    env = gym.make(args.env_name)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 163, in make
    return registry.make(id)
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 119, in make
    env = spec.make()
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/gym/envs/registration.py", line 86, in make
    env = cls(**self._kwargs)
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/simplesim_env.py", line 281, in __init__
    self.multiFBO, self.finalFBO = createFrameBuffers()
  File "/home/daurel_duckietown/Documents/Bachelorarbeit/duckiesim/gym-duckietown/gym_duckietown/envs/simplesim_env.py", line 97, in createFrameBuffers
    True
  File "/home/daurel_duckietown/.local/lib/python3.5/site-packages/pyglet/gl/lib.py", line 105, in errcheck
    raise GLException(msg)
pyglet.gl.lib.GLException: b'invalid operation'

fgolemo, 2018-03-16:
@paulaurel That's the same issue as before: a problem with GL caused by the code using some platform-specific GL instructions that aren't supported by your virtual machine.

As Maxime mentioned here https://github.com/duckietown/gym-duckietown/issues/2#comment-373571330 there is no easy fix but she is working on a solution.

paulaurel, 2018-03-16:
@fgolemo Now I am running it on a PC running ubuntu lts 16.04 not via a virtual machine.

fgolemo, 2018-03-16:
I see. I think the issue is still related to some platform- or driver-specific OpenGL code.
Could you run the command `glxgears` from the Ubuntu package `mesa-utils` and tell us which output you get and if you see any errors? Same for `glxinfo > glxinfo.log` and then please attach (not paste) that text file (`glxinfo.log`).

paulaurel, 2018-03-16:
Here you go. Thanks for your efforts. I am really grateful. 
[glxgears.log](https://github.com/duckietown/gym-duckietown/files/1819443/glxgears.log)
[glxinfo.log](https://github.com/duckietown/gym-duckietown/files/1819444/glxinfo.log)

paulaurel, 2018-03-17:
It works! Thank you!

closed 2018-04-18 by maximecb