duckietown/gym-duckietown #11: Trouble with environment.yaml

liampaull, 2018-05-04:
I got the following output when I ran: `conda env create -f environment.yaml`: 

```
torch 0.3.1.post3 requires pyyaml, which is not installed.
mkl-random 1.0.1 requires cython, which is not installed.
mkl-fft 1.0.0 requires cython, which is not installed.
```

liampaull, 2018-05-04:
not sure how if it's related - but also having trouble with pyglet:

```
ModuleNotFoundError: No module named 'pyglet'
lpaull:gym-duckietown$ sudo pip3 install pyglet
Requirement already satisfied: pyglet in /usr/local/lib/python3.6/site-packages (1.3.0)
Requirement already satisfied: future in /usr/local/lib/python3.6/site-packages (from pyglet) (0.16.0)
lpaull:gym-duckietown$ ./standalone.py --env-name SimpleSim-v0
Traceback (most recent call last):
  File "./standalone.py", line 8, in <module>
    import pyglet
ModuleNotFoundError: No module named 'pyglet'
```
will try to debug

liampaull, 2018-05-04:
right ... obviously have to active the env .. oops. 

now I'm getting an error with cv2:
```
gym-duckietown) lpaull:gym-duckietown$ ./standalone.py --env-name SimpleSim-v0
Traceback (most recent call last):
  File "./standalone.py", line 13, in <module>
    from gym_duckietown.envs import SimpleSimEnv
  File "/home/lpaull/gym-duckietown/gym_duckietown/envs/__init__.py", line 3, in <module>
    from gym_duckietown.envs.duckiebot_env import DuckiebotEnv
  File "/home/lpaull/gym-duckietown/gym_duckietown/envs/duckiebot_env.py", line 10, in <module>
    import cv2
ImportError: /opt/ros/kinetic/lib/python2.7/dist-packages/cv2.so: undefined symbol: PyCObject_Type
```

Seen this one?

maximecb, 2018-05-04:
Never got that one. What's weird here is that that the path says python2.7 (the conda environment specifies Python 3.6). It's also under /opt/ros. So I'm assuming it's an issue with ROS having its own Python install that's conflicting. Normally, the conda environment should prevent that sort of issue, I would think... I hate Python.

Can you do `echo $PYTHONPATH` ? Is `/opt/ros/kinetic/lib/python2.7/dist-packages/` in there?

If so, maybe try: `export PYTHONPATH="`pwd`"`, completely override the Python path before activating the conda environment. Let's see if that works.

maximecb, 2018-05-04:
Sidenote: this is why I try to write software with the least amount of dependencies possible... Because dependencies can be liabilities.

liampaull, 2018-05-04:
worked - closing

closed 2018-05-04 by liampaull

maximecb, 2018-05-04:
Sweet. I was actually just working on a fix. I only use OpenCV to resize frames received from the duckiebot, so I made the opencv import happen lazily (only if you need to run on the duckiebot). I think I can also get rid of that dependency eventually since it's a simple 640x480 => 160x120 resizing (I can just average 4x4 pixel regions using numpy).