duckietown/gym-duckietown #26: Traffic signs

maximecb, 2018-07-06:
I've made a dummy blank street sign mesh using Wings 3D (see `gym_duckietown/meshes/sign_blank.obj`). It's currently not textured, and it doesn't exactly match the dimensions of duckietown signs. We need models and textures to match signs in the [Duckietown appearance specification](https://docs.duckietown.org/opmanual_duckietown/out/duckietown_specs.html#fig:traffic-signs). These should be in obj format and use png textures. Ideally they would have a single texture file per sign.

The signs we most importantly would need are:
- stop
- duckie crossing
- 3 and 4-way intersections

abdelq, 2018-07-06:
Have you looked at https://github.com/duckietown/Software/tree/master/catkin_ws/src/30-localization-and-planning/duckietown_description/urdf/meshes/tags ?

maximecb, 2018-07-06:
No I haven't yet. Are you able to import them into blender and re-export them?

maximecb, 2018-07-14:
Signs now working. Now have support for multiple textures per model. Some material files have the wrong relative file path still though. We should add more signs to the udem1 map to test.

maximecb, 2018-07-14:
@abdelq the circle CI tests are failing when loading the sign_stop image. A little confusing because it works fine on my home machine. Is it possible that the image files you output have the png extension but are not png files?

abdelq, 2018-07-14:
I will look into it. They are actual PNG files:
```
file gym_duckietown/textures/sign_stop.png
gym_duckietown/textures/sign_stop.png: PNG image data, 372 x 748, 8-bit/color RGBA, non-interlaced
```

```
file gym_duckietown/textures/sign_duck_crossing.png
gym_duckietown/textures/sign_duck_crossing.png: PNG image data, 372 x 748, 8-bit/color RGBA, non-interlaced
```

abdelq, 2018-07-14:
@maximecb I'm thinking it's the `..` in the path that causes the issue.

maximecb, 2018-07-16:
Well done Abdelhakim.

closed 2018-07-16 by maximecb