duckietown/gym-duckietown #33: Joystick control script

maximecb, 2018-07-09:
Currently, we have a `manual_control.py` script which allows us to control the duckiebot using the keyboard. It would be nice to have a `joystick_control.py` script which would allow us to control the robot with continuous actions, and to record demonstrations for imitation learning.

The pygame library can be used to get joystick input.

Requirements:
-  The demonstrations should be recorded in the same JSON format as produced by the `experiments/gen_demos.py` script.
- We should support the same gamepad that is used to control duckiebots
- One of the gamepad buttons should start/stop demo recording. Multiple demos can be recorded within a given session.
- Another gamepad button should trigger `env.reset()`

@bhairavmehta95

maximecb, 2018-07-16:
@bhairavmehta95 Good work on the joystick_control script.

Some suggestions to make it even better:

- It's a bit hard to drive in the curves right now. When turning, I lost speed, presumably because it's hard to max out the values on both axes at the same time. Maybe we could use both joysticks, left for throttle, right for steering. It might be smoother?

- When pressing X while recording a demo, the demo should be stopped. The user shouldn't have to stop recording first. It's a detail but it makes recording demos quicker.

- When pressing X, the output file should be rewritten.

- If an episode ends while a demo is being recorded, the demo recording should stop. This is for futureproofing, if we ever end up having a neural net with memory, it might get confused at a demonstration with a restart in the middle.

bhairavmehta95, 2018-07-16:
Will take care of these.

bhairavmehta95, 2018-07-17:
I have a PR ready, but Github won't let me open it so I'll offer my suggestion here @maximecb 

Adding almost all changes requested in #33 .

X button now stops the recording and deletes in no matter what.
If episode finishes when recording, recording is now stopped & saved and the episode resets.

The only thing not added is the double joystick control - usually, double joysticks (one for vel, one for angle)  is better if you can freely move in any axis of movement (i.e a player in a shooting game), but we can't. (Tested this also, and it felt really weird and hardly worked).

I'd suggest if we'd like to do this, we change acceleration to the triggers (forward = Right Trigger, reverse  = Left Trigger), and then can use one joystick for the steering. Haven't played on in a while, but I believe this is how traditional racing games are done as well.

maximecb, 2018-07-18:
@liampaull should the speed of the robot be 0.2m/s when the forward velocity action == 1.0? Is that how the gain calibration parameter works.

closed 2018-07-23 by maximecb