duckietown/gym-duckietown #34: Asymmetric kinematic model

maximecb, 2018-07-09:
The simuator currently uses a simple kinematic model, implemented in the `_update_pos` method of `SimpleSimEnv`. A kinematic model which supports asymmetric wheel dimensions and wheel distances has been developed. It would be desirable to implement this model in the simulator, as it could approximate some of the asymmetric motor behavior observed on real robots.

The new model is described in this short document: https://www.dropbox.com/s/pg9nyysb7pwlici/kinematic_model.pdf?dl=0

abdelq, 2018-07-18:
Will take this.

maximecb, 2018-08-01:
The model is from Ben Weber ETHZ. We should reach out to him as to how to compute the radius of rotation around the Icc.

julian121266, 2018-08-03:
His Github username is: webbe035

julian121266, 2018-08-03:
Quick question: I was playing with the "manual_control.py" script and noticed that the minimal turning radius is very large? Is this by design and matches the radius that Duckiebots have?

maximecb, 2018-08-03:
@julian121266 I also find that the turning radius is quite large.

I tried to directly port the code from here: https://github.com/duckietown/Software/blob/master/catkin_ws/src/05-teleop/dagu_car/src/inverse_kinematics_node.py#L28

This is my version:
https://github.com/duckietown/gym-duckietown/blob/master/gym_duckietown/envs/duckietown_env.py#L18

One thing I did notice is that the robot's speed affects the turning radius. Is this also the case on the real robot? Could you also tell me what is the expected peak speed for the real robot?

julian121266, 2018-08-03:
Hi @maximecb , I don't personally know the right answer to these questions but have relayed it to our robotarium team. Will get back to you when I know more.

abdelq, 2018-08-03:
I believe `0.0318` is fine. Actually the wheels are marketed as having a diameter of `65mm`.

maximecb, 2018-08-03:
@abdelq It's the minimum turning radius, not the wheel radius AFAIK.

abdelq, 2018-08-03:
@maximecb 0.0318m is 3.18cm

maximecb, 2018-08-03:
You're right. But so, is it the minimum Icc turning radius (of the wheel closest to the Icc) or is it the wheel radius?

abdelq, 2018-08-03:
Not sure. I went back to look at the [documentation](//docs.duckietown.org/learning_materials/out/duckiebot_modeling.html). No mention of an \omega_l or \omega_r, only \phi_l and \phi_r. It would be better to wait for the input of others.

julian121266, 2018-08-04:
The robotarium (Julien Kindle) has spoken: 
" (paraphrased)
1. Maximum speed: Nominal 0.22m/s, max I would say 0.5m/s, hard to say. Somewhen above 0.7m/s I think you're unable to steer and only go straight
2. How the turning radius of the Duckiebot is influenced by its speed: The minimal turning radius stays the same, what do you mean?
3. Minimal turning radius: I think ≈ 0.06cm (just a little more such that on-point turns are impossible)"

So a big change is that the turning radius significantly smaller than in the simulation and quite close to 0 for all intents and purposes.

julian121266, 2018-08-06:
@maximecb For robustness purposes it would actually be quite helpful to have the possibility to have different wheel radii. Similar to the way you are already doing domain randomization. 
Other than that am looking forward to much smaller turning radii.

maximecb, 2018-08-06:
@julian121266 

With respect to a smaller turning radius, we need to figure out what is up with the code for the inverse kinematics node. I am seriously wondering whether anyone is using this on the real robot.

As for different wheel radii, @abdelq implemented an asymmetric distance between each wheel and the center of the robot. I think this will produce enough randomization in terms of the robot turning in an asymmetric way.

julian121266, 2018-08-06:
@maximecb Apparently everyone uses that node. Probably it works. What I looked up though is that the simulator script does not use the turning radius of the inverse kinematics node. 

Currently: WHEEL_DIST = 0.102 is used as a turning radius in the simulator script. In "_update_pos()" it is used as a turning radius (l = self.wheel_dist), where "l" is used as a turning radius later on. 

Probably setting "l" to the turning radius should make the Duckiebot go around curves more like the real Duckiebot. 

The asymmetric distance and asymmetric radii have different effect. Asymmetric radii would make the Duckiebot turn when it expects to go straight. You may be right though that this provides plenty of randomization already.

maximecb, 2018-08-06:
I will open a new issue.

closed 2018-08-06 by maximecb