duckietown/gym-duckietown #37: Modularize Code

bhairavmehta95, 2018-07-10:
`simplesim_env.py` is over 1000 lines now, so I propose that we pull some code out into some other files:

* `env_utils.py` - Handles loading of the grid / obstacles from the map file provided
* `agent_utils.py` - Helper functions that help with things regarding the agent (getting and updating current position, heading vectors)
* `collision.py` - Handles everything regarding collisions
* `geometry.py` - Currently, not everything in `collision.py` still has to do with collisions and some functions written for collisions have been repurposed to do other things as well. 

Ideally, we keep `step()`, `render()` (and its helper functions), `reset()`, and `init()` (which can also become a bit more modularized) inside of simplesim, and move everything else out.

maximecb, 2018-07-23:
So, I think we aren't quite going in the direction you suggested, but we did do some refactoring with the object refactoring. One possibility would be to convert the agent itself into a world object class. Might make some sense. Though the agent will always require some special treatment by the simulator.

bhairavmehta95, 2018-08-24:
I'd actually say we've made some good progress in this direction, closing.

closed 2018-08-24 by bhairavmehta95