duckietown/gym-duckietown #45: Traffic light mesh

maximecb, 2018-07-16:
We will eventually want to have traffic lights. For this, we probably need a custom model, because Duckietown uses special red and green traffic lights, no yellow. The model should be low poly (< 1000 polygons). Must have only triangular faces.

@liampaull can you link us an image of that a Duckietown traffic light should look like, for reference?

abdelq, 2018-07-16:
The one I'm using on the Unity simulator comes from https://github.com/duckietown/duckietown-sim-server/tree/master/duckietown_gazebo/models/semaforo

maximecb, 2018-07-16:
@abdelq What does it look like?

abdelq, 2018-07-16:
@maximecb It's pretty basic. A cube with a small protruding circle on each side except the top and bottom, obviously.

liampaull, 2018-07-25:
@tanij do you have a mesh or cad file for the new traffic lights?

closed 2018-10-11 by maximecb