duckietown/gym-duckietown #51: Multiprocessing support for Mac

calvinluo, 2018-07-17:
Currently, Mac users attempts to train an agent using multiple processes with SubprocVecEnv, as is default in main.py, they will come across the following error:

`The process has forked and you cannot use this CoreFoundation functionality safely. You MUST exec().`

From my investigations it seems like the problem occurs when the workers attempt to create their own environment.  Perhaps this is because Mac does not like it when its subprocesses load graphics and figures and such.  Is it possible to bypass this somehow to enable multi-process training?

abdelq, 2018-07-17:
Have you pulled the latest changes?  If so, you can try changing line 39 of `pytorch_rl/vec_env/subproc_vec_env.py` to `set_start_method('spawn')`.
If that's not working too, you are probably better off using docker containers.

calvinluo, 2018-07-17:
It worked, thanks so much!

closed 2018-07-17 by calvinluo

maximecb, 2018-07-17:
Wait, did you need to change the code or was pulling the latest changes sufficient? We should ideally have a solution that works on Mac and linux without changes