duckietown/gym-duckietown #53: Issue with self._drivable_pos(f_pos)

bhairavmehta95, 2018-07-18:
Found by Adam when testing `./manual_control --map_name 4way_adam`

When we hit the 4 way intersection in the middle of the map, `self._drivable_pos(f_pos)` turns to false (I checked all of the conditions in `_valid_pose()`
return (
            self._drivable_pos(self.cur_pos) and # True
            self._drivable_pos(l_pos) and # True
            self._drivable_pos(r_pos) and # True
            self._drivable_pos(f_pos) and # Turns to False
            not self._collision() # True aka hasn't collided with anything.
)

Investigating what the issue is, but if you could also take a look @maximecb that'd be sweet.

bhairavmehta95, 2018-07-18:
Issue is that the 4way tile is not considered to be drivable for some reason.

Happens in both `4way_adam` and `4way`.

maximecb, 2018-07-18:
I would just edit the map files and put /N for now.

maximecb, 2018-07-18:
Maybe we should also have a regression test map for this, `regress_4way`. Could be a map with just a 4way tile, to force the duckie to spawn there, and we do a step with 0 velocity, check that the episode doesn't terminate.

maximecb, 2018-07-18:
I added a little regression test that triggers this bug. Testing is important y'all!

https://github.com/duckietown/gym-duckietown/blob/8de0bba8e3236cef320b31d34c5bb3215ec61402/gym_duckietown/maps/regress_4way_drivable.yaml

maximecb, 2018-07-18:
Sidenote: map with names beginning with `regress_` will now be loaded in the tests but not in the multi map training environment.

closed 2018-07-18 by maximecb