duckietown/gym-duckietown #69: Velocity + steering control

maximecb, 2018-08-06:
@julian121266

I've ported over the code from the [inverse kinematics node](https://github.com/duckietown/Software/blob/master/catkin_ws/src/05-teleop/dagu_car/src/inverse_kinematics_node.py) from the real robot so we could use the same velocity+steering commands as the real robot in the simulator. I had been told that this was the way to implement a minimal turning radius for the robot. However, there have been issues. For one, the rate of turning of the robot is sensitive to the velocity. The minimum turning radius also seems very large.

I've since reviewed the code of the IK node more carefully, and looked at the code in [here](https://github.com/duckietown/Software/tree/master/catkin_ws/src/05-teleop/dagu_car/src). The radius value in the IK code actually seems to refer to the wheel radius, not the turning radius. I actually couldn't find anything related to a minimal turning radius. It also seems that the conversion between velocity and steering to differential drive velocity, the way it's done in the IK node, is inherently sensitive to velocity. The only way to get more sensitive turning is to go at a lower speed.

I'm not sure what to do about this. For now, I will just make the robot in the simulator use smaller velocity values so that it can get more sensitive turning.

julian121266, 2018-08-07:
@maximecb 

Yes, you are right - the radius is actually the wheel radius. Apologies for the confusion. Actually the code implies that the minimal turning radius is 0 now that I think of it (when one wheel is turning one way, and the other equally fast in the other direction). This happens when an omega command as well as a 0 forward velocity command are supplied. I would guess that physics dictates that the faster you are, the larger your turning radius will be. The code in the [gym](https://github.com/duckietown/gym-duckietown/blob/master/gym_duckietown/envs/duckietown_env.py#L46) looks good. 

Please call "angle" [(here)](https://github.com/duckietown/gym-duckietown/blob/master/gym_duckietown/envs/duckietown_env.py#L46) something that reflects that it is an angular velocity. To me it appears that we could close this.

In "simulator.py" in the function "_update_pos()" should the line "w = (Vr - Vl) / l" be "w = (Vr - Vl) / (0.5*l)"?  Maybe I am missing something obvious though.

maximecb, 2018-08-07:
> should the line "w = (Vr - Vl) / l" be "w = (Vr - Vl) / (0.5*l)"? Maybe I am missing something obvious though.

This seems to suggest that it's correct as it is:
http://www8.cs.umu.se/kurser/5DV122/HT13/material/Hellstrom-ForwardKinematics.pdf

closed 2018-08-07 by maximecb

julian121266, 2018-08-07:
Looks good. Thanks for checking.