duckietown/gym-duckietown #73: NameError: name 'DuckietownEnv' is not defined

breandan, 2018-08-09:
Received this error running on Docker on the master branch:

```
root@6d4e86d03b76:/gym-duckietown# python3 pytorch_rl/main.py         --algo a2c         --env-name Duckietown-small_loop-v0         --lr 0.0002         --max-grad-norm 0.5         --no-vis         --num-steps 20
Traceback (most recent call last):
  File "pytorch_rl/main.py", line 298, in <module>
    main()
  File "pytorch_rl/main.py", line 59, in main
    envs = DummyVecEnv(envs)
  File "/gym-duckietown/pytorch_rl/vec_env/dummy_vec_env.py", line 6, in __init__
    self.envs = [fn() for fn in env_fns]
  File "/gym-duckietown/pytorch_rl/vec_env/dummy_vec_env.py", line 6, in <listcomp>
    self.envs = [fn() for fn in env_fns]
  File "/gym-duckietown/pytorch_rl/envs.py", line 15, in _thunk
    basePort = DuckietownEnv.SERVER_PORT
NameError: name 'DuckietownEnv' is not defined
```

maximecb, 2018-08-09:
I think your repo might be out of date?

breandan, 2018-08-09:
Whoops, yep I was using a stale branch.

closed 2018-08-09 by breandan