duckietown/gym-duckietown #83: Coordinate system is not x,y - plane, z -height

julian121266, 2018-09-05:
Hi everyone,

Can we change the coordinate system of the simulator to correspond to a more user friendly coordinate system like x,y create the plane and z is the height (right-hand-rule)? That will greatly accelerate any robotic developments. 

Thank you!

maximecb, 2018-09-05:
If you mean what's internally used in the code, the coordinate system is the OpenGL default. Changing it is possible, but might be pretty tedious. I think it shouldn't be too hard to change what's externally-facing, however. That being said, there are already people using the simulator, and the official start of the competition is getting very close, so I'm not sure I'd recommend going in this direction.

julian121266, 2018-09-06:
Hi Maxime,
Your concerns make sense and I do share them. From where I am sitting it still appears to make sense to at least change the externally-facing coordinates. Currently, I am using coordinate transforms from (z,x,y) to (x,y,z)  plus angle correction to convert what the simulator environment is supplying. Ideally this could be done internally in the simulator such that development of any kind of algorithm using this information is greatly sped up. In the long run the current coordinate system will create more work than changing it and adapting to the more common (x,y,z) coordinate system.

AndreaCensi, 2018-09-06:
@maximecb all the robotics modules use the standard coordinate system convention (e.g. the localization system being developed). I think the external data convention change can be implemented by just a matrix multiplication.

julian121266, 2018-10-11:
Has this issue been addressed?

closed 2019-02-04 by bhairavmehta95