duckietown/gym-duckietown #84: Duckietown - Fully Observable Navigation

bhairavmehta95, 2018-09-09:
This has been a recent request from quite a few people, so I figured I'd give it its own little issue.

gym-duckietown is a quite unique tool in the landscape of simulation, and has many desirable properties of hard, reinforcement learning environments:
- it's a lot faster than many of the physics-based simulators out there
- the visual details are not trivial, but not too extensive to slow it down
- it's got a continuous action space, with the option for discrete actions (that make sense)
- it has the ability to be fully customized (first person vs overhead vs graph-based, obstacles to enable high level planning, reward function design)

One of the requests is that we make a mode for Duckietown to become a fully-observable environment, specifically from a top down view. This would allow for memory-less complex navigation tasks, in addition to the one that will be included in the AIDO release. In addition, since Duckietown itself is a "grid based" world, there has also been interest in making an easier version of the description above, using the traditional grid world physics (from some talks, this was described as useful to test out algorithms that succeed on a standard gridworld, without making too big of a jump to too hard on an environment). 

The implementation would be simple enough; just change the camera position / tracking, using the camera settings from `--draw-curve` as a starting point. 

Ideally, I'd like to develop this as a branch on my fork of the main repo, with the potential for merging if interest goes beyond myself + the few folks who've asked for something like this. Just more of a post to (maybe) garner interest, but also, as a heads up for people who come across this sometime in the future.

maximecb, 2018-09-09:
Cool idea. You might be able to do better than just moving the camera and pointing it down as in the `--draw-curves` mode. There is a `glOrtho` function which you can use in place of `gluLookAt`. It's a camera mode without a field-of-view angle. It implements an orthographic projection, which means everything gets squashed down into a 2D image without perspective distortion.

bhairavmehta95, 2018-09-09:
oh perfect! I will try this out, and keep progress updates posted here!

bhairavmehta95, 2018-09-09:
@eracah You might want to put any specific feature requests here. Just so we can keep everything central.

eracah, 2018-09-10:
ok cool thanks! will do!

fgolemo, 2018-09-11:
@eracah Evan? How come you're working on this too? :D Got your fingers in everything.

eracah, 2018-09-11:
I think the main feature I'd like to see besides some version of "agent_pos" and "get_dir_vec" api from minigrid is an overhead view mask. That is, a (grid_size x grid_size) array where the value of each cell of the array corresponds to what is on the grid (0 for straight road, 1 for the car being in that grid, 2 for a stop sign, ... etc.) or something like that.

That'd be pretty cool! Let me know if that is doable.
Thanks!

eracah, 2018-09-11:
ok perfect, I can discretize the angle to my liking. Thanks!

maximecb, 2018-09-11:
For accessing the map there is a _get_tile(i, j) method which takes integer indices as input. It returns None if there is no tile at that position.

bhairavmehta95, 2018-09-13:
[image]

This is the continuous action scenario. The gridworld-esque one will be similar.

bhairavmehta95, 2018-09-15:
For the folks interested, a manual policy running in the gridworld.
Not sure if we want to have the duckiebot centered in the tile (visually more appealing, but maybe some added randomization?) 

[image]

eracah, 2018-09-18:
This looks really good! Yeah would be cool to toggle between random and centered with a flag.

maximecb, 2018-10-03:
I like the idea of keeping the masks in `info` because Gym and RL implementations have pretty poor support for complex observations that aren't just one n-dimensional tensor.

To decouple things, it might make some sense to have a hierarchy of Renderer classes that can be used to produce the observation and the human render. That way you could parameterize the resolution, anti-aliasing, the camera, etc. for different environments. Just an idea.

I think it makes sense to keep discussions in the main repo if you intend to merge it all back eventually. Otherwise it's two places people have to look.

bhairavmehta95, 2018-10-04:
@maximecb I like the idea of having a hierarchy of renderers - I will sketch something up and maybe run it by this thread before implementing.

Also, is that code opensource yet?

maximecb, 2018-10-05:
@bhairavmehta95 It's private for now just because there isn't very much to show, but I'm hoping to have a basic prototype in 3-4 weeks. I can give you early access if you're interested :)

bhairavmehta95, 2019-02-04:
@maximecb I'm not sure if we ever wanted to merge this into master? Otherwise I will close this issue.

maximecb, 2019-02-04:
Looks like it's already been merged. Seems fine to close the issue.

bhairavmehta95, 2019-02-04:
I was more talking about the functionality of giving the agent the choice to do a gridworld vs get a top down view (the merged only gives the top down view on render rather than obs).

Anyway, if someone needs it, they'll probably stumble upon it. Closing now.

closed 2019-02-04 by bhairavmehta95

pranoy-k, 2019-07-06:
Hi Bhairav, as @maximecb says these changes are merged to master, but I can't see aerial-view option in the main branch(manual_control.py). We are using this gym-environment to create a view from a pedestrian standpoint(the world view must be from pedestrian eyes), Is this possible with the current code, or do we have to undergo a feature implementation?

maximecb, 2019-07-06:
You want to make an environment where pedestrians have to cross the street?

bhairavmehta95, 2019-07-06:
You will need to add a feature implementation. Shouldn't be too hard, you just have to change the position of the camera. Please raise an issue for further discussion.

pranoy-k, 2019-07-08:
@maximecb @bhairavmehta95 
Thank you for the reply

@maximecb We want to control the pedestrians manually to let them cross the street given the autonomous car is roaming on the street. 

Sure @bhairavmehta95 I will raise an issue, it makes more sense to discuss that there