duckietown/gym-duckietown #90: Velocities in simulation and real world do not seem to match

PanagiotisBountouris, 2018-09-29:
I trained a CNN in the simulation using angular and translational velocities, but when I implemented it in the real city I witnessed that I had to multiply the angular velocities by a factor = 4 in order to have similar behavior as in the simulation. Did you face this problem again in the past and do you know why this could happen? Have you compared the outcome of the same velocities (v, ω/omega) in the simulation and in real world in order to be sure that they match?

AndreaCensi, 2018-09-30:
@PanagiotisBountouris sync up with @julian121266 - we might have found some divergences with the model. Also see discussion in #aido-simulator.

maximecb, 2018-09-30:
@bhairavmehta95 do you have the time/bandwidth to take care of the open issues?

bhairavmehta95, 2018-10-01:
Yup
Will take care of it.

bhairavmehta95, 2018-10-12:
Tracking in #81.

closed 2018-10-12 by bhairavmehta95