duckietown/gym-duckietown #111: Cannot solve environment.yaml on conda

catmasteryip, 2018-10-28:
```
(base) C:\Users\Fujits\gym-duckietown>conda env create -f environment.yaml
Solving environment: failed

ResolvePackageNotFound:
  - pytorch=0.3
  - torchvision
```

[image]

I have pytorch 0.4 and pytorch-cpu 0.4 on anaconda, why cant the anaconda prompt resolve the environment? or does it need specifically pytorch 0.3?

maximecb, 2018-10-28:
It should work with pytorch 0.4, but it will print a deprecation warning for some function calls.

catmasteryip, 2018-11-01:
The problem is resolved (temporarily) by commenting the corresponding package requirements in .yaml. But I am not sure if this bypass would back fire later.

liampaull, 2019-02-07:
can we close this?

closed 2019-05-30 by bhairavmehta95