duckietown/gym-duckietown #112: RuntimeError: invalid argument 2: size '[-1 x 4032]'

richielo, 2018-10-28:
Hello, 

I am facing an issue of even running the sample training command 

`python3 pytorch_rl/main.py --no-vis --env-name Duckietown-small_loop-v0 --algo a2c --lr 0.0002 --max-grad-norm 0.5 --num-steps 20`

The error message happens on line 116 in model.py

` x = x.view(-1, 32 * 9 * 14)`

Error message:

`RuntimeError: invalid argument 2: size '[-1 x 4032]' is invalid for input with 127872 elements at ..\aten\src\TH\THStorage.cpp:80`

I am using windows and installed everything with the first provided method using pip. I have not changed anything in the code, looks like size mismatch with the convnet. Thanks for your help in advance

fenjiro, 2018-10-28:
hello,
i had the same problem, after installing gym-duckietown in ubutu 16.06 environement using conda, have you got any feedback or tips to resolve the problem ?

**File "/home/osboxes/gym-duckietown/pytorch_rl/model.py", line 115, in forward
    x = x.view(-1, 32 * 9 * 14)
RuntimeError: invalid argument 2: size '[-1 x 4032]' is invalid for input with 127872 elements at /opt/conda/conda-bld/pytorch_1523244252089/work/torch/lib/TH/THStorage.c:37**

JanFan, 2018-10-28:
Same here on anaconda3-5.2.0 on OSX.

closed 2018-11-06 by bhairavmehta95