duckietown/gym-duckietown #120: Learning of sample program does not converge for python3 pytorch_rl/main.py .......

x15079, 2018-11-26:
Thank you for your help
sorry I understand English a little

--my spec---
gpu gtx 1080
cpu i7-8700k
python3 -- 3.5.2

I had learned the sample program for four days, but did not converge
code[python3 pytorch_rl/main.py --no-vis --env-name Duckietown-small_loop-v0 --algo a2c --lr 0.0002 --max-grad-norm 0.5 --num-steps 20]

No matter how many times the reward will not be over 0.04

please help me

AI-Guru, 2019-01-17:
This bug is very hard to trace. I am currently looking into it.

AI-Guru, 2019-01-17:
Could it be that the agent somehow manages to be on a tile that is not driveable?

This is the tile, when the assertion is violated:
```
{'coords': (1, 3), 'kind': 'asphalt', 'angle': 0, 'drivable': False, 'texture': <gym_duckietown.graphics.Texture object at 0x1329fd0f0>, 'color': array([0.97397976, 0.83174457, 1.14550171])}
```
And this is the position of the agent:
```
[0.61297318 0.         2.43493106]
```

AI-Guru, 2019-01-18:
I did a small workaround. I just end the episode when this error occurs. I am very hesitant to create a pull request. Reason: This is treating the problem's symptoms.

bhairavmehta95, 2019-01-31:
Unfortunately, we haven't been able to find the cause for a few months now. @maximecb I had another user email me about this issue - should we just end the episode? I can't always reproduce this but I can devote some time early next week to finding / fixing.

maximecb, 2019-01-31:
According to the trace, this occurs in a call to `get_agent_info()`, which is called in the `step()` function, after `update_physics()`. In other words, the agent position is updated, and then `get_lane_pos()` is called. I think it's possible that the action puts the agent position outside of a drivable tile.

The cleanest fix IMO would be to add an extra check, call `self._drivable_pos(pos)` in `get_agent_info()` to check that the position is on a drivable tile before calling `get_lane_pos()`.

closed 2019-02-07 by liampaull