duckietown/gym-duckietown #121: Running Headless - Trouble with instructions, cuda gone

AI-Guru, 2019-01-12:
Namaste!

I want to run duckietown headless on AWS, using:
Deep Learning AMI (Ubuntu) Version 20.0 (the newest).

After these steps, something seems to be wrong with CUDA:
```
# Install xvfb
sudo apt-get install xvfb mesa-utils -y

# Remove the nvidia display drivers (this doesn't remove the CUDA drivers)
# This is necessary as nvidia display doesn't play well with xvfb
sudo nvidia-uninstall -y
```

When I execute
```
python -c "import tensorflow"
```
I get
```
Traceback (most recent call last):
  File "/home/ubuntu/anaconda3/envs/tensorflow_p36/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow.py", line 58, in <module>
    from tensorflow.python.pywrap_tensorflow_internal import *
  File "/home/ubuntu/anaconda3/envs/tensorflow_p36/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 28, in <module>
    _pywrap_tensorflow_internal = swig_import_helper()
  File "/home/ubuntu/anaconda3/envs/tensorflow_p36/lib/python3.6/site-packages/tensorflow/python/pywrap_tensorflow_internal.py", line 24, in swig_import_helper
    _mod = imp.load_module('_pywrap_tensorflow_internal', fp, pathname, description)
  File "/home/ubuntu/anaconda3/envs/tensorflow_p36/lib/python3.6/imp.py", line 243, in load_module
    return load_dynamic(name, filename, file)
  File "/home/ubuntu/anaconda3/envs/tensorflow_p36/lib/python3.6/imp.py", line 343, in load_dynamic
    return _load(spec)
ImportError: libcuda.so.1: cannot open shared object file: No such file or directory
...
```

Any ideas?

bhairavmehta95, 2019-01-13:
It might be worth trying without removing the CUDA drivers first. If that doesn't work, we can debug further

bhairavmehta95, 2019-01-17:
Closing for now. If the issue still persists, please reopen.

closed 2019-01-17 by bhairavmehta95