duckietown/gym-duckietown #123: pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"

BlasterL, 2019-01-22:
I was working on my duckietown docker image.
Recently, I reinstall nvidia-docker on my workstation(without internet connection).
Before reinstalling, I backed up my duckietown docker image.
After that, I reinstall nvidia-docker and load my back-up image in my machine.
 I got this error when commnad "python3 run_test.py" in my duckietown docker image.

from ..simulator import Simulator
from pyglet.gl import *
gl._create_shadow_window()
_shadow_window = Window(width=1, height=1, visible=False)
super(XlibWindow, self).__init__(*args, **kargs)
display = get_platform).get_default_display()
return pyglet.canvas.get_display()
return Display()
raise NoSuchDisplayException('Cannot connect to "%s"' % name)
pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"

(These are not full description of my error, sorry.
My  working environment has several separated networks(internet connections) and I cannot use USB driver)

Furthermore "import pyglet.window" in python3 is not working

My duckietown environment was working well before reinstalling but now it is not working.
I don't know reason
Help me :(

bhairavmehta95, 2019-01-31:
Have you run these instructions? https://github.com/duckietown/gym-duckietown#docker-image

Likely, you are forgetting to start Xvfb.

BlasterL, 2019-02-01:
@bhairavmehta95 Thx for your reply.

I ran the instructions you mentioned. Additionally I ran the instructions in "Running headless".

But docker image doesn't have sinter commnad. So I just ran these instructions.

# Add the gym_duckietown package to your Python path
export PYTHONPATH="${PYTHONPATH}:`pwd`"

# Load the GLX library
# This has to be done before starting Xvfb
export LD_LIBRARY_PATH=/Tmp/glx:$LD_LIBRARY_PATH

# Create a virtual display with OpenGL support
Xvfb :$SLURM_JOB_ID -screen 0 1024x768x24 -ac +extension GLX +render -noreset &> xvfb.log &
export DISPLAY=:$SLURM_JOB_ID

After do these, I got the mentioned errors.

BlasterL, 2019-02-07:
@bhairavmehta95  Sorry for late reply because I had national holidays in early this week.

Your instructions makes my environment be working. But it is not completely working.

That means that it prints some texts but cannot visualize the duckietown environment.

It gives me some error, [Errorno 99] about https error.

liampaull, 2019-02-07:
@BlasterL can you file a new issue related to the httpConnection error please?

closed 2019-02-07 by liampaull

BlasterL, 2019-02-21:
Sorry for late reply. 
[image]
[image]

These are full error messages.