duckietown/gym-duckietown #124: AttributeError when training imitation learning model

duckieT, 2019-01-28:
After running ` experiments/train_imitation.py --map-name loop_obstacles`, I got this error:

Traceback (most recent call last):
  File "experiments/train_imitation.py", line 113, in <module>
    model = Model()
  File "experiments/train_imitation.py", line 55, in __init__
    self.apply(init_weights)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/torch/nn/modules/module.py", line 198, in apply
    module.apply(fn)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/torch/nn/modules/module.py", line 198, in apply
    module.apply(fn)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/torch/nn/modules/module.py", line 199, in apply
    fn(self)
  File "/Users/angelwoo/gym-duckietown/experiments/utils.py", line 29, in init_weights
    nn.init.orthogonal_(m.weight.data)
AttributeError: module 'torch.nn.init' has no attribute 'orthogonal_'

Can anyone help :(

bhairavmehta95, 2019-01-31:
What Pytorch version are you using?

bhairavmehta95, 2019-02-04:
Check #130. We're about to add some cleaner IL examples in both TF and Pytorch, so they should be more thoroughly checked than some of the old code. 

Keeping open until we merge.

closed 2019-02-09 by bhairavmehta95