duckietown/gym-duckietown #125: Runtime error when training imitation learning model

duckieT, 2019-01-28:
After running `experiments/train_imitation.py --map-name loop_obstacles`, I got this error:

RuntimeError: size mismatch, m1: [2 x 62016], m2: [2816 x 256 ] at /pytorch/aten/src/THC/generic/THCTensorMathBlas.cu:249

Can anyone please help :(

bhairavmehta95, 2019-01-31:
So the `train_imitation` code is probably quite old, in which case the convolution filters will not produce the same size output as they used to (We changed the image size that comes out of the simulator).

You can take a look at the model getting called and edit the numbers explictly (it should be in the first fully connected layer). I will have some time early next week to fix it otherwise.

bhairavmehta95, 2019-02-04:
Follow in #124 and #130

closed 2019-02-04 by bhairavmehta95