duckietown/gym-duckietown #138: No module named 'utils'

duckieT, 2019-03-02:
The error `ModuleNotFoundError: No module named 'utils'` is returned when trying to run train_imitation.py in 'learning/imitation/tensorflow'

takeitallsource, 2019-03-04:
@bhairavmehta95 do you remember how did you run this code after you moved it here? @liampaull @AndreaCensi are we sure we want this code to be inside the main library repo?

bhairavmehta95, 2019-03-04:
You have to run it from the train imitation script from the `learning/` directory:

`cd learning && python -m imitation.tensorflow.train_imitation`

And yes. This is supposed to be in the main repo. This code has **nothing to do with AIDO** - it's a way for people to use `gym-duckietown` as a library with examples.

closed 2019-03-04 by bhairavmehta95

rhshi, 2019-03-28:
When I run this, it gives me a `ModuleNotFoundError: No module named 'gym_duckietown'` error