duckietown/gym-duckietown #141: Gracefully handle Assert point is not None in get_lane_pos

jrabary, 2019-04-01:
In the function `get_lane_pos` we have the following line

```
def get_lane_pos(self, pos, angle):
        """
        Get the position of the agent relative to the center of the right lane
        """

        # Get the closest point along the right lane's Bezier curve,
        # and the tangent at that point
        point, tangent = self.closest_curve_point(pos, angle)
        assert point is not None
```

This may cause problems for example when training a RL agent with the simultor. Each time the point is None this breaks the whole training pipeline. We need to handle the case `point == None` gracefully.

jrabary, 2019-04-11:
Any ideas on this ?

liampaull, 2019-04-11:
@bhairavmehta95 is the expert on this

liampaull, 2019-04-11:
and most things really :)

bhairavmehta95, 2019-04-11:
Can you pull again from master @jrabary ? This should have been fixed in the latest commits.

closed 2019-04-13 by bhairavmehta95