duckietown/gym-duckietown #143: Obstacles not moving

tejank10, 2019-04-02:
Hello! I tried running `manual_control.py` with `Duckietown-loop_pedestrians-v0`, but the obstacles were not moving. How do I fix this behavior?

liampaull, 2019-04-11:
Sorry for the delay - I think @bhairavmehta95 is the one who might know the answer

bhairavmehta95, 2019-04-11:
Running something like:

`python manual_control.py --env-name=Duckietown-loop_pedestrians-v0 --map-name=loop_pedestrians` 

works.

You have to make sure the map specified has pedestrian duckies with `static` set to `False` in the `.yaml` file.

closed 2019-04-11 by bhairavmehta95

zhaoworking, 2020-01-18:
> Running something like:
> 
> `python manual_control.py --env-name=Duckietown-loop_pedestrians-v0 --map-name=loop_pedestrians`
> 
> works.
> 
> You have to make sure the map specified has pedestrian duckies with `static` set to `False` in the `.yaml` file.

I have already done that according to what u said , but the duckie stll don't moves. How should i do next?

zhaoworking, 2020-01-18:
> Running something like:
> 
> `python manual_control.py --env-name=Duckietown-loop_pedestrians-v0 --map-name=loop_pedestrians`
> 
> works.
> 
> You have to make sure the map specified has pedestrian duckies with `static` set to `False` in the `.yaml` file.

[image]

zhaoworking, 2020-01-18:
I have solved it.