duckietown/gym-duckietown #148: Not compatible with python 3.5+ anymore

takeitallsource, 2019-04-23:
A newly added dependency `dataclasses` have rendered the simulator incompatible with any python version below 3.7.

```
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
```

In python 3.6, we can use `pip install dataclasses` as a workaround but I am not sure if this brings other compatibility issues.

bhairavmehta95, 2019-04-28:
Took the easy way out on this way, fixed in aido2 branch

closed 2019-04-28 by bhairavmehta95

AI-Guru, 2019-05-22:
Unfortunately dataclasses is back. Removing python 3.5 compatibility.

liampaull, 2019-05-24:
@bhairavmehta95 should this issue be re-opened?

bhairavmehta95, 2019-05-24:
I think with the dataclasses, we just need to say its Py3.6 only. It's all a bit too "integrated" to make everything backwards compatible (based on some of the other DT infrastructure stuff)

AI-Guru, 2019-05-26:
It looks to me that dataclasses are used only once in the entire codebase. I would vote for keeping 3.5 compatibility.