duckietown/gym-duckietown #150: Default logging level set to INFO is too verbose

takeitallsource, 2019-04-23:
It is impossible to follow any script output as long as the default logging level for `gym_duckietown` logger is set to `INFO`. 

Do we really need to have that much log output or this is just something we need for debugging AIDO evaluations?

AI-Guru, 2019-06-01:
I second that ;)

There is quite some logging going on. Especially logging the position and angle.

AI-Guru, 2019-06-01:
Insterestingly I did not manage to turn logging of:
```
logging.disable(sys.maxsize)
logger = logging.getLogger()
logger.disabled = True
logger = logging.getLogger("gym-duckietown")
logger.disabled = True
logging.disable(sys.maxsize)
logging.basicConfig(level="ERROR")
```

Any ideas`

robertmoni, 2019-11-13:
Go to ```/gym-duckietown/gym_duckietown/__init__.py``` and change the logging level to your prfecrences.
e.g.:  ```logger.setLevel(logging.WARNING)```

Logging levels:
```
'debug': logging.DEBUG,
 'info': logging.INFO,
 'warning': logging.WARNING,
 'error': logging.ERROR,
 'critical': logging.CRITICAL
```