duckietown/gym-duckietown #151: Syntax error when running manual_control.py

vromerocano, 2019-05-07:
Working on Ubuntu 16.04. When running: 

./manual_control.py --env-name Duckietown-udem1-v0

I get the following:

INFO:gym-duckietown:Registering gym environment id: Duckietown-loop_obstacles-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-udem1-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-small_loop-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-small_loop_cw-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-straight_road-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-loop_empty-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-loop_dyn_duckiebots-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-zigzag_dists-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-loop_pedestrians-v0
INFO:gym-duckietown:Registering gym environment id: Duckietown-4way-v0
Traceback (most recent call last):
  File "manual_control.py", line 16, in <module>
    from gym_duckietown.envs import DuckietownEnv
  File "/home/varomero/gym-duckietown/gym_duckietown/envs/__init__.py", line 4, in <module>
    from gym_duckietown.envs.duckietown_env import DuckietownEnv
  File "/home/varomero/gym-duckietown/gym_duckietown/envs/duckietown_env.py", line 5, in <module>
    from ..simulator import Simulator
  File "/home/varomero/gym-duckietown/gym_duckietown/simulator.py", line 12
    done: bool
        ^
SyntaxError: invalid syntax

bhairavmehta95, 2019-05-07:
What branch are you running?

bhairavmehta95, 2019-05-07:
One of the recent changes we've had is to stop supporting Python 2. Can you let me know what branch / commit you're using + what Python you are running

vromerocano, 2019-05-07:
Hi Bhairav,
I am running the master branch/ commit a0b4e513c3194475e330e76ca3dacb5059b3ad2f
Python 3.5.2

bhairavmehta95, 2019-05-07:
Ah okay - so we've started making changes and utitlizing a feature called `dataclasses`. You can either `pip install dataclasses` in Py3.5, or use >= Py3.6 where the library is standard.

vromerocano, 2019-05-07:
Nice, problem solved. Only additional thing is that I had to pip install pygeometry

closed 2019-05-07 by vromerocano