duckietown/gym-duckietown #160: Missing fontconfig library

maivincent, 2019-06-04:
Issue already raised in #88, but closed because 
> it seems like it might be a unique system thing. 

Someone else already commented to say they had the same problem. I have it too, so it seems it is not an unique system problem.

When running 
>./manual_control.py --env-name Duckietown-udem1-v0

after following the installation process, I get the following message:
```
ImportError: Library "fontconfig" not found.
```

([Full log here](https://pastebin.com/bZMk1Y6H)) 

As suggested in #88 I tried to follow steps from [here](https://unix.stackexchange.com/questions/330068/package-fontconfig-not-found-despite-having-installed-libfontconfig1-dev), without succes.

bhairavmehta95, 2019-06-06:
Please comment with system type, and what Python you are using (virtualenv, conda, standalone and version)

bhairavmehta95, 2019-06-06:
Can you run the steps here: https://unix.stackexchange.com/questions/330068/package-fontconfig-not-found-despite-having-installed-libfontconfig1-dev

And let me know what you see?

maivincent, 2019-06-06:
Thanks @bhairavmehta95 for your answer :)

**System type:** 
Ubuntu 18.04.2 LTS

**Python versions:**

_When I am in the container:_ 
Python 3.6.8 |Anaconda, Inc.| (default, Dec 30 2018, 01:22:34) 
[GCC 7.3.0] on linux

I tried it on both standalone and Conda environments (I installed it in both ways to try).

_In the standalone environment:_
Python 2.7.15rc1 (default, Nov 12 2018, 14:31:15) 
[GCC 7.3.0] on linux2
 
_In the conda environment:_
Python 3.7.1 (default, Dec 14 2018, 19:28:38) 
[GCC 7.3.0] :: Anaconda, Inc. on linux

The error happens the same in all cases.

I'll try the steps and tell you what I see in the next message.

maivincent, 2019-06-06:
"I have found a lot of other resources advertising that this can be fixed by installing both `pkg-config` and `libfontconfig1-dev` "

> $ sudo apt-get install pkg-config libfontconfig1-dev
[sudo] password for vincent: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libfontconfig1-dev is already the newest version (2.12.6-0ubuntu2).
pkg-config is already the newest version (0.29.1-0ubuntu2).
pkg-config set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 126 not upgraded.

"The key here turned out to be the PKG_CONFIG_PATH environment variable. This was empty on a standard shell session on my system."

> $ echo $PKG_CONFIG_PATH
>   _empty_

"There seem to be lots of directories with pkgconfig in the name, but to find the correct one I was able to use apt-file per this thread i.e."

> $ apt-file search fontconfig.pc
 libfontconfig1-dev: /usr/lib/x86_64-linux-gnu/pkgconfig/fontconfig.pc

> $ export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig

> $ echo $PKG_CONFIG_PATH
> /usr/lib/x86_64-linux-gnu/pkgconfig

"And now the ./configure step can find the .pc file which it requires for that library."

> $ nvidia-docker pull duckietown/gym-duckietown && nvidia-docker run -it duckietown/gym-duckietown bash
Using default tag: latest
latest: Pulling from duckietown/gym-duckietown
Digest: sha256:3e20e9739ef286864b56d3211e44d92d09aeb073c1b7758541fd1de993de6bf1
Status: Image is up to date for duckietown/gym-duckietown:latest

> root@1381de2d2d32:/gym-duckietown# Xvfb :0 -screen 0 1024x768x24 -ac +extension GLX +render -noreset &> xvfb.log &
[1] 10
> root@1381de2d2d32:/gym-duckietown# export DISPLAY=:0
> root@1381de2d2d32:/gym-duckietown# ./manual_control.py --env-name Duckietown-udem1-v0
INFO:gym-duckietown:gym-duckietown 2019.0.0

> ...

>Traceback (most recent call last):
  File "./manual_control.py", line 46, in <module>
    env.render()
  File "/gym-duckietown/gym_duckietown/simulator.py", line 1677, in render
    self.speed
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/__init__.py", line 293, in text
    self.document.text = text
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/document.py", line 299, in text
    self.insert_text(0, text)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/document.py", line 430, in insert_text
    self.dispatch_event('on_insert_text', start, text)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/event.py", line 357, in dispatch_event
    if handler(*args):
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/layout.py", line 1031, in on_insert_text
    self._init_document()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/layout.py", line 1020, in _init_document
    self._update()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/layout.py", line 956, in _update
    lines = self._get_lines()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/layout.py", line 932, in _get_lines
    glyphs = self._get_glyphs()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/layout.py", line 1052, in _get_glyphs
    self._document.get_font_runs(dpi=self._dpi),
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/document.py", line 613, in get_font_runs
    ft = self.get_font(dpi=dpi)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/text/document.py", line 623, in get_font
    bold=bool(bold), italic=bool(italic), dpi=dpi)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/__init__.py", line 156, in load
    font = _font_class(name, size, bold=bold, italic=italic, dpi=dpi)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/freetype.py", line 173, in __init__
    self._load_font_face()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/freetype.py", line 192, in _load_font_face
    self._load_font_face_from_system()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/freetype.py", line 195, in _load_font_face_from_system
    match = get_fontconfig().find_font(self.name, self.size, self.bold, self.italic)
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/fontconfig.py", line 397, in get_fontconfig
    _fontconfig_instance = FontConfig()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/fontconfig.py", line 112, in __init__
    self._fontconfig = self._load_fontconfig_library()
  File "/opt/conda/lib/python3.6/site-packages/pyglet/font/fontconfig.py", line 173, in _load_fontconfig_library
    fontconfig = pyglet.lib.load_library('fontconfig')
  File "/opt/conda/lib/python3.6/site-packages/pyglet/lib.py", line 158, in load_library
    raise ImportError('Library "%s" not found.' % names[0])
ImportError: Library "fontconfig" not found.

--> Problem not solved

bhairavmehta95, 2019-06-06:
What version of `pyglet` is installed for you?

bhairavmehta95, 2019-06-06:
Also, let's try to get it working on the standalone, `conda` environment. NVIDIA-Docker brings its own issues, and the other Python you listed won't work (Py2.7)

maivincent, 2019-06-06:
It was probably an NVIDIA-Docker issue. I see that you have removed the instructions about this.
That's good because I think I was confused and I thought it was necessary.

Without getting into the NVIDIA-Docker, it works!

(I just had to pip install pygeometry... even after pulling, and re-creating my conda environment. Can it be that it needs to be added to `environment.yaml`? )

bhairavmehta95, 2019-06-07:
Yup, forgot to add it in there. Thanks again for helping us work through this!

closed 2019-06-07 by bhairavmehta95

yervantk, 2021-05-23:
Still having same problem when running from docker container. Same as above, except running on OSX.

takuma-yoneda, 2022-06-07:
I had the same issue.
In the case of linux (Ubuntu 20.04), just installing it with `apt-get install -y fontconfig` was enough to solve it.
I guess `brew install fontconfig` may just solve it for Mac (https://formulae.brew.sh/formula/fontconfig)?