duckietown/gym-duckietown #162: ImportError: cannot import name 'imresize'

maivincent, 2019-06-10:
When running in gym-duckietown conda environment

`python -m reinforcement.pytorch.train_reinforcement`

I get this error:

```
 File "/home/vincent/gym-duckietown/learning/utils/wrappers.py", line 18, in observation
    from scipy.misc import imresize
ImportError: cannot import name 'imresize'
```

I looked a bit and it seems that `imresize` is deprecated in scipy as written here in _Backwards incompatible changes_:
https://github.com/scipy/scipy/releases/tag/v1.3.0rc1

Could this be the source of the problem?

bhairavmehta95, 2019-06-10:
Yeah likely; let me see if there's a fix for this in the new ones.

bhairavmehta95, 2019-06-10:
[image]

Good catch.

closed 2019-06-10 by bhairavmehta95

maivincent, 2019-06-10:
Fix did induce a new bug: 
```  File "/home/vincent/gym-duckietown/learning/utils/wrappers.py", line 19, in observation
    return np.array(Image.fromarray(observation).resize(shape))
NameError: name 'shape' is not defined
```

maivincent, 2019-06-10:
@bhairavmehta95 :
`return np.array(Image.fromarray(observation).resize([self.shape[0:1]]))`
should be the fix.

bhairavmehta95, 2019-06-10:
Wouldn't it be `self.shape`? The splice (means: from index 0 to 1 exclusive, so just 0) above would resize to a single dimension = square of `shape[0] x shape[0]`

maivincent, 2019-06-10:
Woops, then it should be self.shape[0:2] sorry. It does not accept the 3 args of self.shape, as resize takes in only 2. I figured that shape being defined as  `shape=(120, 160, 3))`, only 120 and 160 are needed.

bhairavmehta95, 2019-06-10:
do0o0o0pe thanks!