duckietown/gym-duckietown #166: Tried Duckietown-loop_pedestrians-v0 but found no obstacles

nitinnat, 2019-06-15:
Hi,

First of all, this is awesome work. I had two questions which I hoped you could answer.

1. I was trying out the manual control option as described in the Usage section of the readme. However, no matter what env_name I provide, I get the same map in the simulator. I am not sure if I am doing something wrong, but when I try Duckietown-loop_pedestrians-v0, I see no moving obstacles, 

2. How would I go about adding another duckiebot in the environment?

I am using Python 3.6 on a Windows machine.

Thanks,
Nitin

nitinnat, 2019-06-18:
Figured out that I had to set the map name as well as env name

nitinnat, 2019-06-18:
Fixed the first one. It would help if someone could tell me how to add another agent.

closed 2019-06-18 by nitinnat

zhaoworking, 2020-01-18:
> Figured out that I had to set the map name as well as env name

I encounter this problem too.So,can u tell me how to set the map name as well as the env name?