duckietown/gym-duckietown #170: scripts.train_cnn.py ModuleNotFoundError

plafer, 2019-06-21:
From the python RL baseline, I ran `python3 -m scripts.train_cnn.py --seed 123`. After the simulation was done, I got the following error message:

>INFO:gym-duckietown:Stopping the simulator because we are at an invalid pose.
/home/plafer/anaconda3/envs/duckietown/bin/python3: Error while finding module specification for 'scripts.train_cnn.py' (ModuleNotFoundError: __path__ attribute not found on 'scripts.train_cnn' while trying to find 'scripts.train_cnn.py')

I'm not sure if it's a known error or not; the file `results/DDPG_123.npz` was successfully created.  However, it's confusing for the user; I'm left wondering if everything worked correctly.

bhairavmehta95, 2019-06-25:
I think this is an issue on the simulator side? I know its a bit late, but do you have any more of the error / stack trace?

bhairavmehta95, 2019-06-25:
Ah nvm, here it is: https://stackoverflow.com/questions/54160948/how-to-fix-the-path-attribute-not-found-error-for-packages-installed-by-pi

`python3 -m scripts.train_cnn.py --seed 123` > python3 -m scripts.train_cnn --seed 123

closed 2019-06-25 by bhairavmehta95