duckietown/gym-duckietown #173: Extrinsic camera calibration

liampaull, 2019-07-03:
It was noted at AI-DO that there is a huge discrepancy in the extrinsic calibration of the camera in the simulator vs. the real robot. 

For a side by side comparison, look where the horizon is in [this simulated vide](http://duckietown-ai-driving-olympics-1.s3.amazonaws.com/v3/testing/by-value/sha256/202beaa343b33249c9f5c8541605dbbe8e1acf6302dc943ec67c69ffd81c2bcf) vs. [this video from a robot in the robotarium](https://duckietown-ai-driving-olympics-1.s3.amazonaws.com/v3/frankfurt/by-value/sha256/056038dd52172f33bbb8b5f52571355ca12edf7296be4b03f7e92890e89977e5)

This huge bias could cause very negative results in sim-to-real transfer I fear.

bhairavmehta95, 2019-07-03:
Yes agreed, we really need to discuss this though, because it's a much bigger bug than it looks - I spent a few weeks on this before AIDO and it was a nightmare to get even this far.

liampaull, 2019-07-04:
yikes

liampaull, 2019-09-04:
@bhairavmehta95 can we have a meeting about this?

liampaull, 2019-09-04:
actually my best guess is that it's not (just) the extrinsic calibration but  that the field of view is wrong for sure 

    CAMERA_FOV_Y = 42

since it's pulled from the camera specs without the fisheye lens

liampaull, 2019-09-04:
I also noticed that the values in `distortion.py` are different than the ones [burned to the vehicle by default](https://github.com/duckietown/duckietown-shell-commands/blob/master/init_sd_card/calib_cam_int_default.yaml)

bhairavmehta95, 2019-09-05:
These are new parameters; they were taken from this file much earlier this year (and much earlier than April, which is when these were pushed).

liampaull, 2019-09-05:
got it - ideally they should be pulled from the same place...

pmirallesr, 2020-08-25:
I just want to raise awareness about the FOV issue. In our own sim2real experiment we have resorted to putting papers under the camera support piece to change its inclination so that it more closely mimicks that of the simulator. This simple trick did a great difference in our sim2real efforts

liampaull, 2020-08-25:
We have a process now of calibrating the camera in the simulator