duckietown/gym-duckietown #175: No module named 'gym-duckietown'

prabhasak, 2019-07-09:
Hi. This error seems to be specific to Windows. I followed suggestions on [adding `PYTHONPATH`](https://stackoverflow.com/questions/3701646/how-to-add-to-the-pythonpath-in-windows-so-it-finds-my-modules-packages) so it finds the gym_duckietown module, but that didn't help either (The learning algorithms are running perfectly on Ubuntu and Mac). Can someone help me with this? Thanks!

_Originally posted by @prabhasak in https://github.com/duckietown/gym-duckietown/issues/139#issuecomment-509410746_

liampaull, 2019-07-10:
We don't officially support Windows at this point, but I will leave open to see if someone finds a fix

bhairavmehta95, 2019-07-10:
Unfortunately we do not officially support windows, but your best bet is to use anaconda. Can you use that and tell me how it goes (Anaconda with Python 3.6 or 3.7)

bhairavmehta95, 2019-07-17:
Closing, we have been able to confirm working with Anaconda on Windows. Unfortunately, we will not be supporting any other configurations on Windows.

closed 2019-07-17 by bhairavmehta95