duckietown/gym-duckietown #194: How to test my own RL algorithm in dukcietown with the graphics.

zhaoworking, 2020-01-17:
I have suceessfully passed the Test "./manual_control.py --env-name Duckietown-udem1-v0".
But I wanna test my own RL algorithm. So can u tell me how to do that?
Should i creat a file first? and run it in the simulator. And how should i make a virtual display in which i can observe how the car runs.In other words, i wanna test my RL algorithm wit the graphics in the duckuietown.

zhaoworking, 2020-03-13:
please help me , i get stuck in it.

liampaull, 2020-03-13:
Hi - can you take at look at the AIDO RL baseline. The vanilla RL one with DDPG is [here](https://docs.duckietown.org/daffy/AIDO/out/embodied_rl.html), and there's a new residual policy learning baseline (beta) [here](https://docs.duckietown.org/DT19/AIDO/out/embodied_rpl.html). Let us know if those are useful.

zhaoworking, 2020-03-14:
thank u, but why can't i see my simulator when i am testing my RL with DDPG. How do i display my simulator when i test it.

zhaoworking, 2020-03-14:
> Hi - can you take at look at the AIDO RL baseline. The vanilla RL one with DDPG is [here](https://docs.duckietown.org/daffy/AIDO/out/embodied_rl.html), and there's a new residual policy learning [...]

thank u, but why can't i see my simulator when i am testing my RL with DDPG. How do i display my simulator when i test it.

zhaoworking, 2020-03-14:
> thank u, but why can't i see my simulator when i am testing my RL with DDPG. How do i display my simulator when i test it.
I have found the file called 'manual_control.py', and i can see the simulator and map when i was testing .but when i test the initial file 'train_reinforcement.py' , i can't see that.

zhaoworking, 2020-03-14:
> Hi - can you take at look at the AIDO RL baseline. The vanilla RL one with DDPG is [here](https://docs.duckietown.org/daffy/AIDO/out/embodied_rl.html), and there's a new residual policy learning [...]

I have found the file called 'manual_control.py', and i can see the simulator and map when i was testing .but when i test the initial file 'train_reinforcement.py' , i can't see that.

closed 2020-03-14 by zhaoworking