duckietown/gym-duckietown #204: The simulator's position is too small when testing with RL algorithm.

zhaoworking, 2020-03-14:
`timestep: 487 | reward: 2.9815825234448745
INFO:gym-duckietown:Pos: [0.16776178 0.         0.29571088] angle 4.266275815950335
INFO:gym-duckietown:Pos: [0.18177636 0.         0.27306946] angle 4.049985441742995
timestep: 488 | reward: 3.515600246424384
INFO:gym-duckietown:Pos: [0.18177636 0.         0.27306946] angle 4.049985441742995
INFO:gym-duckietown:Pos: [0.20296474 0.         0.24407919] angle 4.112490319974118
timestep: 489 | reward: 3.9223733389333497
INFO:gym-duckietown:Pos: [0.20296474 0.         0.24407919] angle 4.112490319974118
INFO:gym-duckietown:Pos: [0.2160342  0.         0.22504895] angle 4.10870028266937
timestep: 490 | reward: 3.8976755001671903
INFO:gym-duckietown:Pos: [0.2160342  0.         0.22504895] angle 4.10870028266937
INFO:gym-duckietown:Pos: [0.20268746 0.         0.24788006] angle 4.258060757087562
timestep: 491 | reward: 3.624382276087298
INFO:gym-duckietown:Pos: [0.20268746 0.         0.24788006] angle 4.258060757087562
INFO:gym-duckietown:Pos: [0.18788337 0.         0.27996628] angle 4.3021555331039965
timestep: 492 | reward: 3.378131035184803
INFO:gym-duckietown:Pos: [0.18788337 0.         0.27996628] angle 4.3021555331039965
INFO:gym-duckietown:Pos: [0.17724695 0.         0.30111433] angle 4.190612199495656
timestep: 493 | reward: 3.0978186555579086
INFO:gym-duckietown:Pos: [0.17724695 0.         0.30111433] angle 4.190612199495656
INFO:gym-duckietown:Pos: [0.18037936 0.         0.29412895] angle 4.391082117996021
timestep: 494 | reward: 2.886240731778871
INFO:gym-duckietown:Pos: [0.18037936 0.         0.29412895] angle 4.391082117996021
INFO:gym-duckietown:Pos: [0.18032057 0.         0.29688446] angle 4.9910317739651795
timestep: 495 | reward: 2.896586904763131
INFO:gym-duckietown:Pos: [0.18032057 0.         0.29688446] angle 4.9910317739651795
INFO:gym-duckietown:Pos: [0.1788154  0.         0.29126345] angle 4.957022812380346
timestep: 496 | reward: 2.8599158783976595
INFO:gym-duckietown:Pos: [0.1788154  0.         0.29126345] angle 4.957022812380346
INFO:gym-duckietown:Pos: [0.17792745 0.         0.27865244] angle 4.608344250004112
timestep: 497 | reward: 3.0078651414030046
INFO:gym-duckietown:Pos: [0.17792745 0.         0.27865244] angle 4.608344250004112
INFO:gym-duckietown:Pos: [0.17331981 0.         0.29421553] angle 4.240753406541636
timestep: 498 | reward: 3.0340761428067067
INFO:gym-duckietown:Pos: [0.17331981 0.         0.29421553] angle 4.240753406541636
INFO:gym-duckietown:Pos: [0.17084038 0.         0.30560628] angle 4.755370910399716
timestep: 499 | reward: 2.976090021658111
`
Why my postion is so small until the end of the tset ? I tested with this [file.](https://github.com/duckietown/gym-duckietown/blob/master/learning/reinforcement/pytorch/train_reinforcement.py)

zhaoworking, 2020-03-14:
@Bhairav Mehta

zhaoworking, 2020-03-14:
@AndreaCensi

zhaoworking, 2020-03-14:
why my speed is 0 ? it is spinning aroud all the time.

AndreaCensi, 2020-03-15:
@CourchesneA ?

bhairavmehta95, 2020-03-15:
This might just be an RL thing, where the agent hasn't trained enough yet.

zhaoworking, 2020-03-16:
> This might just be an RL thing, where the agent hasn't trained enough yet.
but why my speed is zero , it is just spinning at one place. should I increase the steps ? And how long will it be trained enough?

zhaoworking, 2020-03-16:
> This might just be an RL thing, where the agent hasn't trained enough yet.

This might just be an RL thing, where the agent hasn't trained enough yet.
but why my speed is zero , it is just spinning at one place. should I increase the steps ? And how long will it be trained enough?

zhaoworking, 2020-03-16:
`INFO:gym-duckietown:Pos: [0.10907588 0.         0.08145748] angle -1.4140188871210604
INFO:gym-duckietown:Pos: [0.1077757  0.         0.06491242] angle -1.5707276305686322
timestep: 1229 | reward: 0.5541583695478534
INFO:gym-duckietown:Pos: [0.1077757  0.         0.06491242] angle -1.5707276305686322
INFO:gym-duckietown:Pos: [0.1072284  0.         0.06348076] angle -0.8405789910143403
DEBUG:gym-duckietown:No tile found at [ 0.14709685  0.         -0.00442732] (0, -1)
DEBUG:gym-duckietown:Invalid pose. Collision free: True On drivable area: False
DEBUG:gym-duckietown:safety_factor: 1.0
DEBUG:gym-duckietown:pos: [0.09121964 0.         0.04560005]
DEBUG:gym-duckietown:l_pos: [ 0.14709685  0.         -0.00442732]
DEBUG:gym-duckietown:r_pos: [0.03534243 0.         0.09562742]
DEBUG:gym-duckietown:f_pos: [0.15125248 0.         0.1126527 ]
**INFO:gym-duckietown:Stopping the simulator because we are at an invalid pose**.
timestep: 1230 | reward: -10
INFO:gym-duckietown:Pos: [0.1072284  0.         0.06348076] angle -0.8405789910143403
Total T: 1230 Episode Num: 8 Episode T: 82 Reward: 388.986034
DEBUG:gym-duckietown:No tile found at [0.2196841  0.         0.59513809] (0, 1)
DEBUG:gym-duckietown:Invalid pose. Collision free: True On drivable area: False`

And I always encounter this interruption , leaving the simulator be stopped.

zhaoworking, 2020-03-16:
> @CourchesneA ?

INFO:gym-duckietown:Pos: [0.10907588 0. 0.08145748] angle -1.4140188871210604
INFO:gym-duckietown:Pos: [0.1077757 0. 0.06491242] angle -1.5707276305686322 timestep: 1229 | reward: 0.5541583695478534 INFO:gym-duckietown:Pos: [0.1077757 0. 0.06491242] angle -1.5707276305686322 
INFO:gym-duckietown:Pos: [0.1072284 0. 0.06348076] angle -0.8405789910143403 
DEBUG:gym-duckietown:No tile found at [ 0.14709685 0. -0.00442732] (0, -1) 
**DEBUG:gym-duckietown:Invalid pose. Collision free: True On drivable area: False** 
DEBUG:gym-duckietown:safety_factor: 1.0 
DEBUG:gym-duckietown:pos: [0.09121964 0. 0.04560005] 
DEBUG:gym-duckietown:l_pos: [ 0.14709685 0. -0.00442732]
DEBUG:gym-duckietown:r_pos: [0.03534243 0. 0.09562742] 
DEBUG:gym-duckietown:f_pos: [0.15125248 0. 0.1126527 ] 
**INFO:gym-duckietown:Stopping the simulator because we are at an invalid pose. timestep: 1230 | reward: -10** 
INFO:gym-duckietown:Pos: [0.1072284 0. 0.06348076] angle -0.8405789910143403 
Total T: 1230 Episode Num: 8 Episode T: 82 Reward: 388.986034 
DEBUG:gym-duckietown:No tile found at [0.2196841 0. 0.59513809] (0, 1) 
DEBUG:gym-duckietown:Invalid pose. Collision free: True On drivable area: False

And I always encounter this interruption , leaving the simulator be stopped. Then i have to wait for a long time for the next training procedure.

zhaoworking, 2020-03-16:
> This might just be an RL thing, where the agent hasn't trained enough yet.

oh , i see .thank u. But can u take a look at the interruption problem.

zhaoworking, 2020-03-16:
> This might just be an RL thing, where the agent hasn't trained enough yet.

oh , i see. And could u take a look at the interruption problem?

bhairavmehta95, 2020-03-16:
It's not an interruption, it's the training of the algorithm that takes a long time, especially if you're working on a laptop or CPU based hardware. Not much we can do about that.

zhaoworking, 2020-03-16:
> It's not an interruption, it's the training of the algorithm that takes a long time, especially if you're working on a laptop or CPU based hardware. Not much we can do about that.

Does it mean that i need a GPU on my laptop , and then it will process faster?

closed 2020-03-19 by zhaoworking