duckietown/gym-duckietown #228: Gym lane tiles not in line with duckietown specification

pmirallesr, 2020-08-24:
I was having a look at the stock tiles that come when you install gym-duckietown, and I noticed they do not follow the duckietown specification. Here is the spec for reference
[image]

And here is one of the curves used in duckietown

[image]

The curvature is massively different, resulting in way more driving space in the stock tiles than in the duckumentation spec. This difference not only makes replication harder when doing sim2real (breaking the whole purpose of having an appearance specification) but also hides a fundamental conflict between the specs and how gym-duckietown is implemented:

The simulator only detects out of lane situations when the cart physically moves out of a driveable tile. With the normal tiles this is not an issue as mostly, going beyond the white line = going out of the tile and so the violation of the driving rules is correctly punished with an environment restart. However, if your tiles follow the duckietown specification, the cart is free to move beyond the white line while staying within the tile, making it harder to train agents to stay within the line.

As a side note, one of the default maps included in gym-duckietown, zigzag_dists, violates the topological constraints regarding lane adjacency. This is a limiting constraint on its own, reducing the variety of legal maps that can be built with a small set of tiles.

liampaull, 2020-08-25:
great point! @CourchesneA do you know how to easy update the tile images in the sim?

CourchesneA, 2020-08-25:
Since we do not actually use the white lane from the texture, I think that it isn't such a bad thing that is it closer to the border of the tile since it is closer to what actually happens (i.e. record out-of-bound if we drive to the next tile outside the curve). Until we actually implement a mechanism to check for in-tile out of lane (i.e. for rewards or penalties) I think a good solution is to use the following texture, which was already in duckietown:

[image]
(https://github.com/duckietown/gym-duckietown/blob/master/gym_duckietown/textures/curve_left_2.png)

This is still not exactly respecting the duckietown specs, but it is inbetween simulator behavior and real-world appearance. Here's the PR: https://github.com/duckietown/gym-duckietown/pull/229

closed 2020-08-26 by liampaull