duckietown/gym-duckietown #240: TypedDict from typing library

nerkn, 2020-12-06:
Recently my sim failed to work in google colab. Error is this, any suggestions?

/usr/local/lib/python3.6/dist-packages/duckietown_world/world_duckietown/map_loading.py in <module>()
     14 from .duckiebot import DB18
     15 from .duckietown_map import DuckietownMap
---> 16 from .old_map_format import MapFormat1Object
     17 from .other_objects import (
     18     Barrier,

/usr/local/lib/python3.6/dist-packages/duckietown_world/world_duckietown/old_map_format.py in <module>()
----> 1 from typing import Dict, List, NewType, TypedDict, Union
      2 
      3 __all__ = ["MapFormat1", "MapFormat1Object", "MapFormat1Constants"]
      4 
      5 

ImportError: cannot import name 'TypedDict'

mostafaelaraby, 2020-12-06:
@nerkn  I recommend you to check [this Colab notebook ](https://github.com/duckietown/challenge-aido_LF-baseline-dagger-pytorch/blob/main/notebook.ipynb ) that used the gym-duckietown to train an imitation learning baseline, 
check the dependencies installations and the virtual display setup

nerkn, 2020-12-06:
Same error, It doesnt work either 
>  File "/usr/local/lib/python3.6/dist-packages/duckietown_world/world_duckietown/old_map_format.py", line 1, in <module>
>    from typing import Dict, List, NewType, TypedDict, Union
> ImportError: cannot import name 'TypedDict'

https://colab.research.google.com/github/duckietown/challenge-aido_LF-baseline-dagger-pytorch/blob/main/notebook.ipynb

greta-p, 2020-12-06:
The problem here is python 3.6.

for < 3.8 need a library called `typing_extensions` installed