duckietown/gym-duckietown #243: I am sure this project can not run, can you fix the bugs?

Fanshaoliu, 2021-04-08:
I tried different platform(win, linux ,macos) to install your project, and different ways for installing, but noone works,

frank-qcd-qk, 2021-04-08:
Can you illustrate the issue in more detail? I have just recently installed and it works. Can you provide how you have installed it?

Have you tried these instructions?

`git clone https://github.com/duckietown/gym-duckietown.git
cd gym-duckietown
pip3 install -e .`

Fanshaoliu, 2021-04-08:
Thanks for your answer, I have tried the instructions, but I meet similar problems in the three platform:

```
DEBUG:gym-duckietown:Falling back to non-multisampled frame buffer
DEBUG:gym-duckietown:Falling back to non-multisampled frame buffer
Traceback (most recent call last):
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/run_tests.py", line 39, in <module>
    env = DuckietownEnv(map_name=map_name)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/envs/duckietown_env.py", line 16, in __init__
    Simulator.__init__(self, **kwargs)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 334, in __init__
    self._load_map(map_name)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 756, in _load_map
    self.map_file_path = get_resource_path(f"{map_name}.yaml")
  File "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/resources.py", line 86, in get_resource_path
    raise ZKeyError(msg, known=sorted(res2), res1=sorted(res1))
zuper_commons.types.exceptions.ZKeyError: Could not find resource 'small_loop_only_duckies.yaml'.
```

but infact 'small_loop_only_duckies.yaml' is exist, when I fixed bugs of "can't find resource", new bugs will be key error about "colors"

Fanshaoliu, 2021-04-08:
> Can you illustrate the issue in more detail? I have just recently installed and it works. Can you provide how you have installed it?
> 
> Have you tried these instructions?
> 
> `git clone https://github.com/duckietown/gym-duckietown.git`
> 
> `cd gym-duckietown`
> 
> `pip3 install -e .`

Thanks for your answer, I have tried the instructions, but I meet similar problems in the three platform:

DEBUG```
:gym-duckietown:Falling back to non-multisampled frame buffer
DEBUG:gym-duckietown:Falling back to non-multisampled frame buffer
Traceback (most recent call last):
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/run_tests.py", line 39, in <module>
    env = DuckietownEnv(map_name=map_name)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/envs/duckietown_env.py", line 16, in __init__
    Simulator.__init__(self, **kwargs)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 334, in __init__
    self._load_map(map_name)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 756, in _load_map
    self.map_file_path = get_resource_path(f"{map_name}.yaml")
  File "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/resources.py", line 86, in get_resource_path
    raise ZKeyError(msg, known=sorted(res2), res1=sorted(res1))
zuper_commons.types.ex
```ceptions.ZKeyError: Could not find resource 'small_loop_only_duckies.yaml'.

but infact 'small_loop_only_duckies.yaml' is exist, when I fixed bugs of "can't find resource", new bugs will be key error about "colors"

frank-qcd-qk, 2021-04-08:
A Few things:
1. the gym duckietown is supposedly supported only on Ubuntu 20.04
2. as for the map, the map dependencies are from the duckietown world. The supported map file names are here: https://github.com/duckietown/duckietown-world/tree/daffy/src/duckietown_world/data/gd1/maps

Fanshaoliu, 2021-04-08:
> A Few things:
> 
> 1. the gym duckietown is supposedly supported only on Ubuntu 20.04
> 2. as for the map, the map dependencies are from the duckietown world. The supported map file names are here: https://github.com/duckietown/duckietown-world/tree/daffy/src/duckietown_world/data/gd1/maps

Thank you for you reminding, I tried to replace my "duckietown_world/data" with the git you provided, now I meet new bugs

`DEBUG:commons:version: 6.1.7 *
INFO:typing:version: 6.1.8
DEBUG:duckietown_world:duckietown-world version 6.2.7 path /Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages
INFO:geometry:version: 2.0.5
DEBUG:gym-duckietown:gym-duckietown version 6.1.2 path /Users/liushaofan/PycharmProjects/gym-duckietown/src

INFO:gym-duckietown:Information about the graphics card:
 information: dict[4]
              │ vendor: ATI Technologies Inc.
              │ renderer: AMD Radeon Pro 5300M OpenGL Engine
              │ version: 2.1 ATI-3.9.15
              │ shading-language-version: 1.20
DEBUG:gym-duckietown:Falling back to non-multisampled frame buffer
DEBUG:gym-duckietown:Falling back to non-multisampled frame buffer
INFO:duckietown_world: data: /Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/data
.yaml.yaml
/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/data/gd1/maps/udem1.yaml
DEBUG:gym-duckietown:loading map file "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/data/gd1/maps/udem1.yaml"
INFO:gym-duckietown:done
DEBUG:gym-duckietown:loading mesh 'duckiebot' from file_path '/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/data/gd1/meshes/duckiebot/duckiebot.obj'
DEBUG:gym-duckietown:loading materials from /Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/duckietown_world/data/gd1/meshes/duckiebot/duckiebot.mtl
WARNING:gym-duckietown:could not find material 'gkmodel0_chassis_geom0_mat_001-material' in ['']
WARNING:gym-duckietown:could not find material 'gkmodel0_chassis_geom0_mat_001-material.001' in ['']
Traceback (most recent call last):
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 841, in _interpret_map
    self.mesh = get_duckiebot_mesh(default_color)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 1997, in get_duckiebot_mesh
    return get_mesh("duckiebot", change_materials=change_materials)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/objmesh.py", line 41, in get_mesh
    mesh = ObjMesh(old_file_path, mesh_name, segment, change_materials)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/objmesh.py", line 166, in __init__
    chunks[-1]["end_idx"] = len(faces)
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/run_tests.py", line 10, in <module>
    env = gym.make("Duckietown-udem1-v0")
  File "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/gym/envs/registration.py", line 145, in make
    return registry.make(id, **kwargs)
  File "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/gym/envs/registration.py", line 90, in make
    env = spec.make(**kwargs)
  File "/Users/liushaofan/opt/anaconda3/envs/safe-rl/lib/python3.8/site-packages/gym/envs/registration.py", line 60, in make
    env = cls(**_kwargs)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/envs/duckietown_env.py", line 16, in __init__
    Simulator.__init__(self, **kwargs)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 334, in __init__
    self._load_map(map_name)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 763, in _load_map
    self._interpret_map(self.map_data)
  File "/Users/liushaofan/PycharmProjects/gym-duckietown/src/gym_duckietown/simulator.py", line 856, in _interpret_map
    raise InvalidMapException(msg, map_data=map_data)
gym_duckietown.exceptions.InvalidMapException: Cannot load map data
map_data:
dict[3] 
│ tiles:
│ list[7]
│ #0 [floor, floor, floor, floor, floor, floor, floor, floor]
│ #1 [floor, curve_left/W, straight/W, 3way_left/W, straight/W, straight/W, curve_left/N,
│     asphalt]
│ #2 [floor, straight/S, grass, straight/N, asphalt, asphalt, straight/N, asphalt]
│ #3 [floor, 3way_left/S, straight/W, 3way_left/N, asphalt, asphalt, straight/N, asphalt]
│ #4 [floor, straight/S, grass, straight/N, asphalt, curve_right/N, curve_left/E, asphalt]
│ #5 [floor, curve_left/S, straight/E, 3way_left/E, straight/E, curve_left/E, asphalt,
│     asphalt]
│ #6 [floor, floor, floor, floor, floor, floor, floor, floor]
│ objects:
│ dict[14]
│ │ tree1: {kind: tree, pos: [2.5, 4.5], rotate: 180, height: 0.25, optional: True}
│ │ duckie1: {kind: duckie, pos: [2.5, 2.9], rotate: -90, height: 0.08, optional: True}
│ │ sign1: {kind: sign_stop, pos: [2.08, 4.05], rotate: 90, height: 0.18}
│ │ sign2: {kind: sign_left_T_intersect, pos: [0.94, 3.96], rotate: 90, height: 0.18}
│ │ sign3: {kind: sign_stop, pos: [2.08, 2.96], rotate: -90, height: 0.18}
│ │ sign4: {kind: sign_right_T_intersect, pos: [0.94, 3.05], rotate: -90, height: 0.18}
[... cut ...]
│ │ house1: {kind: house, pos: [4.8, 2.6], rotate: 90, height: 0.5, optional: True}
│ │ truck1: {kind: truck, pos: [1.6, 6.3], rotate: 0, height: 0.2, optional: True}
│ │ bus1: {kind: bus, pos: [2.0, 0.5], rotate: 0, height: 0.18, optional: True}
│ │ bus2: {kind: bus, pos: [4, 0.5], rotate: 0, height: 0.18, optional: True}
│ │ bus3: {kind: bus, pos: [6.0, 0.5], rotate: 0, height: 0.18, optional: True}
│ │ truck3: {kind: truck, pos: [7.5, 0.5], rotate: 180, height: 0.2, optional: True}
│ tile_size: 0.585`

Fanshaoliu, 2021-04-08:
Question is solved by:

```
git clone -b master https://github.com/duckietown/gym-duckietown.git

cd gym-duckietown

pip3 install -e .

```

The default branch is bad😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭😭, it waste too much time to find this

closed 2021-04-08 by frank-qcd-qk