duckietown/gym-duckietown #248: Repo examples are out of date and have several issues

matwilso, 2021-05-04:
I wanted to mess around with imitation learning on a simple lane following expert. Based on the README, I thought this would be easy to test out. But I had to edit several parts of the code, like to deal with python 3.8 features, and to account for renaming of directories. And even when I got things running, the line following agent didn't work. I had to revert back to an older commit and then it did work.

I could have messed up the install or missed something. But that is the experience I had trying to get something working from this repo. And based on other active issues, I don't think I am entirely alone.

If anyone is still actively maintaining this repo, there is work to be done to make it better to run out of the box. Or remove mention of things that don't work anymore, and fix broken links and references to renamed directories (e.g., pytorch --> basic)

amirhesamyazdi, 2021-06-15:
I want to do the same thing but I face a lot of issues. 
I used docker hub installation but I face ImportError: Library "fontconfig" not found.
and after I try some suggested fixes by Duckietown collaborators it still same error. I can't use pytorch_rl too even though pytorch is installed and I confirmed it by importing torch successfully.

Would you please share exactly how you got it working? I and I'm sure many would appreciate your input.

tamas-visy, 2021-10-31:
I was able to solve the `ImportError: Library "fontconfig" not found`:

You can install it via `apt-get install libfontconfig-dev` (make sure to call `apt-get update` and maybe `apt-get upgrade` before).