duckietown/gym-duckietown #257: The standalone Dockerfile does not work.

AlemSnyder, 2021-08-28:
https://github.com/duckietown/gym-duckietown/blob/a6df727604bbe69dff0a50ba2a375f66e1c730c5/docker/standalone/Dockerfile#L6

I think there is supposed to be a backslash, or maybe this should be a requirements file.

adleris, 2023-04-19:
Commit #273 fixes that layer in the compilation by adding in the '/'.

After fixing that, I had some other issues getting compilation working. I then get 

```
ERROR [6/6] RUN python3 -c "from gym_duckietown import *"
...
ModuleNotFoundError: No module named 'gym_duckietown'
```

Skipping that layer, I tried running `python3 manual_control.py`, and I had some other issues I couldn't resolve.

1. I had `ModuleNotFoundError: No module named 'gym'`, which then worked when I added a `pip3 install gym`.
2. After installing that, I had `pyglet.canvas.xlib.NoSuchDisplayException: Cannot connect to "None"` 

I've tried looking through some previous versions of the `docker/standalone/Dockerfile` and had other issues getting them to build, so I stopped my investigation.

closed 2023-04-19 by liampaull

adleris, 2023-04-20:
@liampaull there were still some issues with the dockerfile that I mentioned in that comment, should that be a separate issue?

commit 7bd97246 (2023-04-19): Merge pull request #273 from adleris/fix/257-dockerfile-compilation-fix

#257: fix dockerfile compilation