duckietown/gym-duckietown #262: texture loading is failing on Windows environment

etnifsilva, 2022-01-15:
I tried to run the command that is on docs below:
`./manual_control.py --env-name Duckietown-udem1-v0`

but a error was aways been arrising:
```
Traceback (most recent call last):
  File "C:\dev\pos\duckietown\gym-duckietown\manual_control.py", line 35, in <module>
    env = DuckietownEnv(
  File "C:\dev\pos\duckietown\gym-duckietown\src\gym_duckietown\envs\duckietown_env.py", line 16, in __init__
    Simulator.__init__(self, **kwargs)
  File "C:\dev\pos\duckietown\gym-duckietown\src\gym_duckietown\simulator.py", line 381, in __init__
    self.reset()
  File "C:\dev\pos\duckietown\gym-duckietown\src\gym_duckietown\simulator.py", line 638, in reset
    fn = get_texture_file(f"tiles-processed/{self.style}/{kind}/texture")[0]
  File "C:\dev\pos\duckietown\gym-duckietown\venv\lib\site-packages\duckietown_world\world_duckietown\map_loading.py", line 306, in get_texture_file
    raise ZKeyError(msg, tried=tried)
zuper_commons.types.exceptions.ZKeyError: Could not find any texture for tiles-processed/photos/floor/texture
tried:
[tiles-processed/photos/floor/texture.jpg, tiles-processed/photos/floor/texture.png, 
 tiles-processed/photos/floor/texture.tga, tiles-processed/photos/floor/texture.bmp, 
 tiles-processed/photos/floor/texture_1.jpg, 
 tiles-processed/photos/floor/texture_1.png, 
 tiles-processed/photos/floor/texture_1.tga, 
 tiles-processed/photos/floor/texture_1.bmp, 
 tiles-processed/photos/floor/texture_2.jpg, 
 tiles-processed/photos/floor/texture_2.png, 
 tiles-processed/photos/floor/texture_2.tga, 
 tiles-processed/photos/floor/texture_2.bmp, 
 tiles-processed/photos/floor/texture_3.jpg, 
 tiles-processed/photos/floor/texture_3.png, 
 tiles-processed/photos/floor/texture_3.tga, 
 tiles-processed/photos/floor/texture_3.bmp, 
 tiles-processed/photos/floor/texture_4.jpg, 
 tiles-processed/photos/floor/texture_4.png, 
 tiles-processed/photos/floor/texture_4.tga, tiles-processed/photos/floor/texture_4.bmp]

Process finished with exit code 1

```
I debbuged the code and saw this line of code:
`fn = get_texture_file(f"tiles-processed/{self.style}/{kind}/texture")[0]`
https://github.com/duckietown/gym-duckietown/blob/7c06d1ef9671ebfefc019e9607753345f2d13407/src/gym_duckietown/simulator.py#L638

this shoud work well on linux machines, but in a Windows machine it's rising a error, like showed before.
I replaced the line with error to this line of code:
`fn = get_texture_file(os.path.join("tiles-processed", self.style, kind, "texture"))[0]`

and the error was gone.

this is the commit from my fork:

https://github.com/etnifsilva/gym-duckietown/pull/1/commits/50ce1349644c68e1240ffbd35f07990d9bad472c

uchansansan, 2022-02-10:
same problem(

Snoup97, 2022-06-25:
thx, had the same issue. the following line solved it for me.
`fn = get_texture_file(os.path.join("tiles-processed", self.style, kind, "texture"))[0]`