duckietown/gym-duckietown #264: Install CUDA, PyTorch or any DNN Library on the duckiebot itself?

bishoyroufael, 2022-04-24:
I have a DB21M assembled, getting confused about how to run neural networks on the bot itself. I tried SSHing into the bot but didn't find any NVIDIA drivers installed or CUDA installed? Is there a way to do that easily? Can't find anything useful in the docs talking about this.

From the NVIDIA docs, there should be a way to install JetPack and get some neural networks running out from their examples presented [here](https://github.com/dusty-nv/jetson-inference). I want to use that with ROS on the duckiebot

FelixMildon, 2022-05-18:
Can this please be explained @tanij , this understanding would help allot.

bishoyroufael, 2022-05-18:
After going through a lot of research and pain. I was able to make something work [here](https://github.com/bishoyroufael/duckie-dqlpp-docker) as part of my thesis project. Feel free to use the `Dockerfile` in your own project. 

It uses  `dustynv/jetson-inference` as a base container which is basically a container having CUDA, PyTorch and some cool DNN models there to be used straight away. It also has ROS `melodic` on top of that where you can set `ROS_MASTER_URI` with the duckiebot IP for communication to work.

If you just want PyTorch you can create rather derive from `l4t-pytorch`. More details about that [here](https://catalog.ngc.nvidia.com/orgs/nvidia/containers/l4t-pytorch).