duckietown/gym-duckietown #265: env.render_obs(True) error TypeError: unhashable type: 'list'

GreatGameDota, 2022-07-20:
Environment is Ubuntu 20.04.4 with Python 3.8.10 and gym-duckietown-daffy==6.1.30

Minimum to reproduce
```py
from duckietown_gym import DuckietownEnv

env = DuckietownEnv()

env.reset()
env.render()
seg = env.render_obs(True)
```

Full stack trace
```
DEBUG:gym-duckietown:loading materials from /home/usr/.local/lib/python3.8/site-packages/duckietown_world/data/gd1/meshes/signs/sign_generic/sign_generic.mtl
Traceback (most recent call last):
  File "test.py", line 7, in <module>
	seg = env.render_obs(True)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/simulator.py", line 1958, in render_obs
	observation = self._render_img(
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/simulator.py", line 1907, in _render_img
	obj.render(draw_bbox=self.draw_bbox, segment=segment, enable_leds=self.enable_leds)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/objects.py", line 147, in render
	self.render_mesh(segment, enable_leds=enable_leds)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/objects.py", line 69, in render_mesh
	self.mesh.render(segment=segment)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/objmesh.py", line 362, in render
	self = get_mesh(self.mesh_name, True)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/objmesh.py", line 48, in get_mesh
	mesh = ObjMesh(old_file_path, mesh_name, segment, change_materials)
  File "/home/usr/.local/lib/python3.8/site-packages/gym_duckietown/objmesh.py", line 276, in __init__
	texture = load_texture(fn2, segment=segment, segment_into_color=segment_into_color)
TypeError: unhashable type: 'list'
```

This error occurs on gym-duckietown-daffy>=6.0.55.

The error occurs with different meshes based on the random starting position of the duckiebot.

liampaull, 2022-08-26:
@Velythyl do you have any idea about this? I know we were using segmented images for the object detection exercise

Velythyl, 2022-09-14:
Yeah I am pretty sure I had a PR about this at some point but I don't know what happened to it. I can take a look