duckietown/gym-duckietown #280: No module named 'numpy.testing.utilis'

fourb1, 2024-02-22:
I try to follow this procedure: [gym-duckietown](https://github.com/duckietown/gym-duckietown)

I try to launch the Duckietown simulation environment with the following command (I work with Ubuntu 22.04.3):

`./manual_control.py --env-name Duckietown-udem1-v0`

I get the following result:

```
Traceback (most recent call last):
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/geometry/manifolds/special_euclidean_group.py", line 76, in belongs
    self.SOn.belongs(R)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/geometry/manifolds/special_orthogonal_group.py", line 39, in belongs
    assert_allclose(det, 1, err_msg="I expect the determinant to be +1.")
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/geometry/utils/numpy_backport.py", line 14, in assert_allclose
    from numpy.testing.utils import assert_array_compare  # @UnresolvedImport
ModuleNotFoundError: No module named 'numpy.testing.utils'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/duckie/gym-duckietown/./manual_control.py", line 35, in <module>
    env = DuckietownEnv(
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/gym_duckietown/envs/duckietown_env.py", line 16, in __init__
    Simulator.__init__(self, **kwargs)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/gym_duckietown/simulator.py", line 381, in __init__
    self.reset()
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/gym_duckietown/simulator.py", line 755, in reset
    self.state = p.initialize(c0=c0, t0=0)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/duckietown_world/world_duckietown/dynamics_delay.py", line 31, in initialize
    state0 = self.factory.initialize(c0, t0, seed)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/duckietown_world/world_duckietown/pwm_dynamics.py", line 61, in initialize
    return DynamicModel(self, c0, t0, 0.0, 0.0)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/duckietown_world/world_duckietown/pwm_dynamics.py", line 135, in __init__
    GenericKinematicsSE2.__init__(self, c0, t0)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/duckietown_world/world_duckietown/generic_kinematics.py", line 26, in __init__
    geo.SE2.belongs(q0)
  File "/home/duckie/dt-sim/lib/python3.10/site-packages/geometry/manifolds/special_euclidean_group.py", line 79, in belongs
    raise ValueError(msg)
ValueError: The rotation is not a rotation:
[[-0.7803981  -0.62528298]
 [ 0.62528298 -0.7803981 ]]
```

I tried to install different versions of Numpy but no change.

I went to see the file python "/home/duckie/dt-sim/lib/python3.10/site-packages/gym_duckietown/envs/duckietown_env.py" and this is what it looks like:

```
# coding=utf-8
from numpy.ma.core import allclose

import numpy as np

__all__ = ["assert_allclose", "check_allclose"]

try:
    from numpy.testing.utils import assert_allclose  # @UnusedImport
except ImportError:

    def assert_allclose(actual, desired, rtol=1e-7, atol=0, err_msg="", verbose=True):
        """ Backporting assert_allclose from Numpy 1.5 to 1.4 """
        from numpy.testing.utils import assert_array_compare  # @UnresolvedImport

        def compare(x, y):
            return np.allclose(x, y, rtol=rtol, atol=atol)

        actual, desired = np.asanyarray(actual), np.asanyarray(desired)
        header = "Not equal to tolerance rtol=%g, atol=%g" % (rtol, atol)
        assert_array_compare(compare, actual, desired, err_msg=str(err_msg), verbose=verbose, header=header)

def check_allclose(actual, desired, rtol=1e-7, atol=0, err_msg="", verbose=True):  # @UnusedVariable
    if not allclose(actual, desired, rtol=rtol, atol=atol):
        err_msg += "\n%s != %s " % (actual, desired)
        raise Exception(err_msg)
```

kcala461, 2024-04-01:
Hi, I have the same problem, have you solved it?

fourb1, 2024-04-01:
> Hi, I have the same problem, have you solved it?

Yes, my problem was with the Numpy version. I used the following command and it solved my problem : 

`pip install numpy==1.22.1`

fourb1, 2024-04-01:
My problem was with the Numpy version. I used the following command and it solved my problem :

`pip install numpy==1.22.1`

closed 2024-04-01 by fourb1